uavpal/disco4g #88: Glympse

bato2000, 2021-03-29:
Hi, I'm Ivan from SpainThank you very much for the work with the 4g mod of the parrot disco.
I have noticed a problem with glympse. 
It does not give a link to track the drone, it only gives the glympse page
Many thanks for everything

s0ftice, 2021-03-29:
Hi, it's a known issue caused by Glympse, it hasn't been working for about 4 weeks. We informed them but no reaction so far.

Same thing happened about a year ago. Took them two weeks to fix it...

s0ftice, 2021-04-05:
Glympse should be working again in the latest beta (see details here: https://uavpal.com/disco/faq#beta). Or simply replace the uavpal_glympse.sh file using https://raw.githubusercontent.com/uavpal/disco4g/master/disco/uavpal/bin/uavpal_glympse.sh

closed 2021-04-05 by s0ftice

skyer29, 2021-11-10:
> Glympse should be working again in the latest beta (see details here: https://uavpal.com/disco/faq#beta). Or simply replace the uavpal_glympse.sh file using https://raw.githubusercontent.com/uavpal/disco4g/master/disco/uavpal/bin/uavpal_glympse.sh
I tried to replace glympse.sh file in BIN folder by filezilla but doesn’t work.

s0ftice, 2021-11-10:
can you elaborate on "filezilla doesn't work"? what part of filezilla does not work? can you send some screenshots?

skyer29, 2021-11-10:
no, filezilla works fine.
I changed the glympse.sh file in BIN folder using filezilla.
I have to do other things?

s0ftice, 2021-11-10:
you can run `ulogcat |grep uavpal` 1 or 2 minutes after the modem goes online and send me the output here.

andresfg1, 2021-11-23:
[image]

s0ftice, 2021-11-23:
you have not set a API key in the glympse config file...

andresfg1, 2021-11-23:
Right, it was my fault, I'm going to do the test correctly.

andresfg1, 2021-11-23:
thanks, using the latest beta version, glympse messages work fine

[image]

skyer29, 2021-12-02:
after numerous attempts to install and remove I finally solved the glympse problem.  I installed the latest version of the softmod and then I overwrote the glympse.sh file taken from the beta version of bebop2. So it works good.

joaquinfranco1, 2025-01-23:
Right now I try to go to Glympse https://developer.glympse.com/ , and I cannot sign in into the Glympse, and I cannot get the link.

If I go to register my app: https://developer.glympse.com/account/create its a error 404 in the website. 

does anyone knows how to get the link code now? Thanks.

joaquinfranco1, 2025-01-23:
[image]

I cannot get the Glympse link key:

[image]
[image]
[image]

s0ftice, 2025-01-24:
please see here https://github.com/uavpal/disco4g/issues/114 how to manually request for an API key from Glympse.