uavpal/disco4g #92: 2021 Huawei modem for disco & parrot softmod

parrot333, 2021-05-21:
Hi,

I'm new to RC plane. I decide to start it using parrot disco. Very interested in disco because we can use 4G as well. There are some questions:

1. About 4G modem. I live in Indonesia. And it's difficult to get huawei e3372h-153, mostly they sell e8372 - 608 which i'm not sure it can be used or not (haven't found any review about this). I guess new huawei model always come and i'm a bit late to fly disco in 2021 😂 maybe somebody can give me any solution?

2. If i wanna do softmod for parrot disco. Which should i start first? Update latest parrot software & skycontrol 2 before softmod? Or do softmod first and then update parrot & skycontrol 2 software?

Thank you

parrot333, 2021-05-22:
*Sorry, mostly they sell e8372H 607

s0ftice, 2021-05-22:
1. You can find other compatible models here https://uavpal.com/disco/faq#othermodems The E8372h-608 should be fine. Only few reports of a newer modem firmware that doesn't work nicely. Check out aliexpress.com, should not be a problem finding a E3372h-153. Just make sure you don't get a -320!

2. First upgrade the Parrot firmware on Disco and Skycontroller 2, then do the softmod installation.

Feel free to join us on Slack: https://uavpal.com/slack

closed 2021-05-22 by s0ftice