uavpal/disco4g #93: Parrot Disco - RTH error, automatic landing but no return home.

Robifpv20, 2021-06-14:
Parrot Disco - błąd RTH, automatyczne lądowanie lecz bez powrotu do domu... Witajcie panowie i panie i wszyscy miłośnicy podniebnych lotów Parrot Disco. Dużo godzin mam przelatane lecz głównie na oryginalnym wi-fi, a tylko parę na LTE.  I właśnie kilka dni temu spotkało mnie to czego obawia się każdy pilot drona. Dron przestał reagować, przestał wykonywać polecenia i sam zaczął lądować ruchem okrężnym, ale uwaga: nie w domu, nie w miejscu startu, lecz 200 metrów ode mnie. Straciłem kontrolę nad nim 200 metrów ode mnie i włączył sobie sam  obniżanie lotu i wylądował u sąsiada na drzewie. Czy ktoś miał coś takiego ? Jedynie szczęście takie, że straty nie są duże, a dron i kamera jest do naprawienia.

Polish - English translation
Parrot Disco - RTH error, automatic landing but without returning home ... Hello, ladies and gentlemen, and all lovers of Parrot Disco sky flights. I have flown many hours, but mainly on the original wi-fi, and only a few on LTE. And just a few days ago, what every drone pilot is afraid of happened to me. The drone stopped responding, stopped following commands and started landing by itself in a circular motion, but beware: not at home, not at the take-off site, but 200 meters from me. I lost control of it 200 meters away from me and it triggered the descent itself and landed at the neighbor's tree. Has anyone had something like this? Only luck is that the losses are not large, and the drone and the camera can be repaired.

Robifpv20, 2021-06-14:
I would like to add that during the landing with a circular, circular motion, the camera image was not tilted, it was horizontal as during RTH. The difference is that the drone, instead of turning back, began to land immediately. All this happened during a gentle lowering of the flight and a slight turn. As if he was suddenly ordered to land in a remote place. The battery was over 4V per cell. Before this landing, the connection was very poor, it was not steered well. And the picture was sometimes tilted differently than I was turning ... I have a videoe screen recorder from this flight. It is very strange. The drone very often asks for recalibration after switching to 4G. I will also say that I like to fly with a 720P view and I had it set up then too. It was good only at the beginning, after a few minutes the image was still there, but the control was very delayed.

Robifpv20, 2021-06-14:
...but the control was very delayed.

ClassAxion, 2021-06-14:
Were you flying normally (manual) before losing range? You didn't do anything automatic like orbiting? Send us a video (you can upload on youtube as unpublic) that starts a few moments before the disco parrot loses signal.

Robifpv20, 2021-06-14:
Ok. I will send you a movie recorded on a smartphone as unlisted, because this movie in the modem on the micro SD card is damaged and cannot be played. As for this flight, I was controlling it manually all the time. The last command to fly was probably the right stick towards myself, so I ordered him to fly up, because I noticed that he was starting to fly down, a moment earlier I performed a few commands to the left and right. As far as I remember, I did not give him the command to orbit and wait in the air, neither right nor left.

ClassAxion, 2021-06-14:
You can try playing the damaged file in VLC: https://www.videolan.org/vlc/download-windows.pl.html

Robifpv20, 2021-06-14:
please, here's a video from the phone:

https://youtu.be/5cKwQ5YcXaI

ClassAxion, 2021-06-14:
It doesn't looks like RTH because you saw it on the app. I don't see any RTH messages and you don't get disconnected from the disco.

s0ftice, 2021-06-14:
Two things, check if your issue is not the same as this https://github.com/uavpal/disco4g/issues/64

Secondly, circular landing on the spot (not at the take-off locations) happens only if the drone loses GPS signal, which almost never happens. Check that you have not placed anything on top of the front part of your CHUCK.

Robifpv20, 2021-06-14:
> Nie wygląda jak RTH, ponieważ widziałeś to w aplikacji. Nie widzę żadnych komunikatów RTH i nie rozłączasz się z dyskoteką.

I'll call you later, I have to go to work. Let me just say it was very weird because the video was horizontal like RTH and I couldn't do anything even though I tried to lift it up as it taxied down.

Robifpv20, 2021-06-14:
> Dwie rzeczy, sprawdź, czy Twój problem nie jest taki sam jak ten #64
> 
> Po drugie, lądowanie okrężne w miejscu (nie w miejscach startu) ma miejsce tylko wtedy, gdy dron traci sygnał GPS, co prawie nigdy się nie zdarza. Sprawdź, czy nie umieściłeś niczego na przedniej części uchwytu CHUCK.

That's how I read this whole thread with problem number 64. It is possible that Chuck crashed, blocked and did not want to execute commands, or the commands were so delayed that he was executing the previous commands from a few - a dozen or so seconds ago. As for the GPS, there is a Huawei modem above the CHUCK, glued to the cover and ends in front of the word Parrot. So I believe that the GPS is not obstructed.

Robifpv20, 2021-06-14:
Please watch carefully from 6:45 am. I turn left and the image is tilted right, then I turn right and the image is left. It looks very unnatural that the camera is looking in a different direction than the direction of flight. As if I was giving the camera turn commands, not the drone. I was watching the camera all the time, but at the end it was as if I wasn't controlling it. As if the disco picked up some disturbance and obeyed the command, as in the flight plan. Forgive the blurry image, but after the last landing, the camera needs to be cranked up and focus. Can a blurry camera image affect flight?

Robifpv20, 2021-06-14:
Here is a video of Wi-Fi flights, the image is only blurry at the beginning and then it improves halfway.  But that's not a problem, the camera's already cut, I can fix it.  What worries me the most is the emergency landing when using 4G, because nothing like that happened on Wi-Fi.

https://youtu.be/9bil25klJnY

Robifpv20, 2021-06-14:
As a curiosity, I will tell you that in the garden of this gentleman where I landed Disco, there has been another drone for a month now, which also landed without returning home. Chinese version of the Mavic Pro - SG906 pro. Could someone have a signal jammer in this area, or something is emitting very high noise in this area? 
I already have a lot of flights in this area using wifi and there have never been problems. It's only the first time.

Robifpv20, 2021-06-14:
I found the reason !!! there is a fault, won't you believe it?

ClassAxion, 2021-06-14:
So what's the problem?

Robifpv20, 2021-06-14:
The left servo mechanism is very hard to turn down. As if it was obliterated. You can pull it up and turn it, but it's really hard to go down. I don't know if it happened in flight or maybe from falling from a tree. After turning on the voltage, both servos seem to level, but when I am doing a flight simulation, the right servo works fine, and the left one only up and does not come back down anymore. If the servo overheated in flight, then we have the cause of the failure and automatic landing.

ClassAxion, 2021-06-15:
It makes sense. Maybe these are fake emax servos?

Robifpv20, 2021-06-15:
Zamieniłem wtyczki serwa lewego i prawego. Wszystko wskazuje na to że lewe serwo jest uszkodzone, ciezko chodzi. Po przesmarowaniu i dodaniu oliwy, jest lekka poprawa. Ale i tak lewe serwo się zacina. 99 % że powodem awarii była blokada serwa.

ClassAxion, 2021-06-15:
Super! I was afraid that it was a software problem after all. 

I don't know if disco can diagnose servo failure like it does with motor failure (if so, an alert should pop up), however, looking at the video, I find that the C.H.U.C.K was trying to take control of the machine, so the image was similar to RTH. 

This was not a controller-initiated landing, the parrot just dropped by itself.

Robifpv20, 2021-06-16:
I also believe Chuck has taken control and initiated an emergency landing.  Although you can see in the last seconds that I was adding gas and tried to raise Disco up.  I believe Chuck is checking the servos and detecting if any of them are sticking.  The right good servo when starting Disco makes one move to level, and the left bad servo makes two moves to be horizontal, first down and then back up.  During simulated take-off, the right servo moves up and down nicely, while the left servo only goes up and stops.  The Disco has an autopilot that supports it when the speed and roll is too low, so that it does not stall.  As in the final landing phase, we can steer left and right, but we can't go up and down anymore.  I also thought it was software or GPS fault, but luckily it's not Chuck's fault, but a broken servo.  And now very important !!!  Listen: I used these servos for 20 hours, and the previous owner said that I was 40 hours, which is 60 hours in total !!!  They should be replaced much earlier!  sorry for my poor english but i try my best to provide you with valuable information to fly safely and without crashing our favorite plane called Disco :)

Robifpv20, 2021-06-16:
Please check if Pushbullet works in your free time. Because it doesn't work for me. I get an incomplete link, no traceability. Pushbullet only sends a link to the main Glympse website

s0ftice, 2021-06-16:
see here: https://github.com/uavpal/disco4g/issues/88 (fix for Glympse)

Robifpv20, 2021-06-16:
OK, thank you s0ftice. I will have to learn how to transfer files to Disco and replace that file.

Robifpv20, 2021-06-16:
Everything is correct, the servo is broken and jams. I changed the servo plugs again and the left one stops after a while, when it stops, it stops working and stays in the down position, causing the drone to taxi to the right and the autopilot activates the automatic landing. Although the drone is in the starting position and CHUCK commands the ailerons up, the left one goes up and works for a short time. Later it goes down and is permanently locked in this position. I tried to push the aileron handle up, but I felt that the motor was working all the time and was holding it down, and the right aileron was still working properly and moving. I've also heard that it sometimes happens that the motor windings break where the silicone cables connect to the thin winding wires. Apparently, this winding breaks right next to the engine, so I put a small amount of glue under the cable and over the cable so that it does not move, so that it does not vibrate from vibrations and hard landings.

[url=https://postimg.cc/grDWH975][img]https://i.postimg.cc/grDWH975/IMG-20210616-154622.jpg[/img][/url]

Telepilote, 2021-06-17:
In the video it looked like the disco went into a flat spin so the servo
fault makes lots of sense. 👍

Robifpv20, 2021-06-17:
Servos replaced and adjusted. They work very well. Flight simulation done, everything indicates that the plane is ready to fly. A short test flight tomorrow, but only on wi-fi for now. When replacing the servo, pay attention to leveling and properly screw the aileron handles. If they are not equally even on both sides, if the left servo is higher and the right servo is lower, then maybe I got two right servos. You have to unscrew one servo and remove the main shaft with a rack, turn it 180 degrees and put it back. After this, both servos will stop at the same height.

Robifpv20, 2021-06-18:
Today was the first test after the last crash, all systems are operational, the disco goes straight and soars up nicely, it's nice. It was a bit windy so I didn't go crazy too long, not too far. After 10 minutes, I landed gently, without comment. I guess all right, only now came my servo tester. I'll have fun with it tomorrow :)

closed 2021-08-19 by s0ftice

Telepilote, 2021-11-01:
---------- Forwarded message ---------
From: benjamin jones ***@***.***>
Date: Mon, 1 Nov 2021 at 18:43
Subject: Re: Invoice 1011 from Jones Conrete & Construction
To: Peter Hofland ***@***.***>

Thank you guys, have received.

Ben jones.

On Sun, 31 Oct 2021, 9:43 pm Peter Hofland, ***@***.***>
wrote:

> Hi Ben,
> Deposit sent, coming from ING so may take a day or two depending on who
> you bank with. Talk again soon.
>
> Regards, Peter & Diana.
>
> On Sun, 31 Oct 2021 at 16:17, Jones Concrete & Construction <
> ***@***.***> wrote:
>
>>
>>
>> INVOICE NO. 1011 DETAILS
>> Jones Concrete & Construction
>>
>>
>> DUE 28/10/2021
>> A$3,000.00
>> Review and pay
>> <http://links.notification.intuit.com/ls/click?upn=LEV65WI9EZ1l5TkUt4hKqzq6J49VEXJMRhUxqRckK3UK6eeEy0A-2FylQ0WpPN9IR9nATq3WhntsTLMCi919IHWM-2FH2OegTOm0j-2FRUFYCSJMSRzU-2By41IBhb4XC7AkbE8yap0B-2Bk-2B4e0SFlNrsfG7mNDZ9WPtUERKhWXMGV-2BgiLgkbuqrtIDKrtZkOEQ0znKJYjRH9q6mYhKBoYAB2lkgyX9bJqNp6RTGnPdMhqdVlOthcJgdEHZs6uYV-2FytdK5R9fig0y06OjHU1v9aAcZmL7Bg-3D-3D8fMu_bQiQ5-2BGmXnymGnBlJujf8tkwj-2BB5QDQ7IdQrLCC6j4u57rNnTY0XSZIlfrED7bMJEvsf4ZBvsN-2B-2Ff7gED6-2BE1iq6BDzNUH0CaI9-2FJ6-2BfpYlQ6UFYctK37YOfP31QEB7uB-2FsNC-2FkjHtugpiUzwdaQc7iRp7k-2FAhuAPgOV5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXIUCd4WCbpBu330BP3ff4df-2Bmezn0I7IKlSpyfKphRBCppS0l4sRUnSacfrLc7z6Pn5zNXsvrjxvI9zw9oCBsDHQR7T5y8y6-2Bdl0h60C-2BPZKotZcZMVwu-2B-2Fd0h5UfITz9z-2BK-2FaBIDxo0pAmCDjY8-2Fif6vtg-2BDfVBJ9X0AklXzfbfcLrBhoRxGIlF31CJMwUpW5a-2F7Gt5bueIJg-2FM-2B3go76MVUTbzFViBzqq1aGYCBD3gmkvjgEnNG1DdIlmpjctyo21l9IrlN-2F2l4XCV7n8t34wHck3HLUr-2FqWR-2BA0PDEApQzUy7TzD5IQ7w-2FVeKItHzxzAGssYlzHy2KiCfyvyEnMwGIdCNEIlQSZTo-3D>
>> Powered by QuickBooks
>>
>> Dear Peter Hofland,
>>
>> Here's your invoice! We appreciate your prompt payment.
>>
>> Have a great day,
>> Jones Conrete & Construction
>> Invoice to
>> Peter Hofland
>> Peter
>> A$3,000.00
>> Deposit for concrete works and excavation
>> 1 X A$3,000.00
>> Includes GST total A$272.73
>> Total A$3,000.00
>> Balance due A$3,000.00
>> Review and pay
>> <http://links.notification.intuit.com/ls/click?upn=LEV65WI9EZ1l5TkUt4hKqzq6J49VEXJMRhUxqRckK3UK6eeEy0A-2FylQ0WpPN9IR9nATq3WhntsTLMCi919IHWM-2FH2OegTOm0j-2FRUFYCSJMSRzU-2By41IBhb4XC7AkbE8yap0B-2Bk-2B4e0SFlNrsfG7mNDZ9WPtUERKhWXMGV-2BgiLgkbuqrtIDKrtZkOEQ0znKJYjRH9q6mYhKBoYAB2lkgyX9bJqNp6RTGnPdMhqdVlOthcJgdEHZs6uYV-2FytdK5R9fig0y06OjHU1v9aAcZmL7Bg-3D-3DZHEW_bQiQ5-2BGmXnymGnBlJujf8tkwj-2BB5QDQ7IdQrLCC6j4u57rNnTY0XSZIlfrED7bMJEvsf4ZBvsN-2B-2Ff7gED6-2BE1iq6BDzNUH0CaI9-2FJ6-2BfpYlQ6UFYctK37YOfP31QEB7uB-2FsNC-2FkjHtugpiUzwdaQc7iRp7k-2FAhuAPgOV5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXIUCd4WCbpBu330BP3ff4df-2Bmezn0I7IKlSpyfKphRBCppS0l4sRUnSacfrLc7z6Pn5zNXsvrjxvI9zw9oCBsDHQR7T5y8y6-2Bdl0h60C-2BPZKotZcZMVwu-2B-2Fd0h5UfITz9z-2BK-2FaBIDxo0pAmCDjY8-2Fif6vtg-2BDfVBJ9X2BV-2FGScAjh9ut0haKx9A7k2VNL2IqJo1aCHZfMGoWlKwyEh1jzw4ZFCfA9xXtDRvPNe-2BWnhDYMP2ebnbDmd7q1zcBsrz54FKivWPcjmTutmromS7-2F-2F8LK0T63CBpXVoQM7Oa507zNAly9yqr7z0tg9DUZMMuhDf-2FeNXnrtJttXaTWZZDO1y-2FLWJCqGjmtaRQ4-3D>
>>
>> Jones Concrete & Construction
>> ***@***.***
>> ABN 25 428 093 011
>> If you receive an email that seems fraudulent, please check with the
>> business owner before paying.
>>
>> <http://links.notification.intuit.com/ls/click?upn=LEV65WI9EZ1l5TkUt4hKqx7hpHgOSLDS3Nfk-2BVe7nJOKWgKKJw-2Bvz2cuWMx1lAEWyNpLS35fbyZ9xA5LU5LlLS5hCQshv4rEzfRCpDoaDJ-2FFcGrI6caNrVtUJj-2FgCSvA1-Fl_bQiQ5-2BGmXnymGnBlJujf8tkwj-2BB5QDQ7IdQrLCC6j4u57rNnTY0XSZIlfrED7bMJEvsf4ZBvsN-2B-2Ff7gED6-2BE1iq6BDzNUH0CaI9-2FJ6-2BfpYlQ6UFYctK37YOfP31QEB7uB-2FsNC-2FkjHtugpiUzwdaQc7iRp7k-2FAhuAPgOV5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXIUCd4WCbpBu330BP3ff4df-2Bmezn0I7IKlSpyfKphRBCppS0l4sRUnSacfrLc7z6Pn5zNXsvrjxvI9zw9oCBsDHQR7T5y8y6-2Bdl0h60C-2BPZKotZcZMVwu-2B-2Fd0h5UfITz9z-2BK-2FaBIDxo0pAmCDjY8-2Fif6vtg-2BDfVBJ9X1knFeZNLp-2F-2B2fFrI5BLQKajH1jZ9r6weiQ1PwzgwbsnIg4AZrhnlGLUes34pt5mJ8iSkO02BEMyIs6qljEWfWbyLznFUAcay9CRQJqaOJIFQKAZ5RiNMITOlccduz8o9NuhPTv-2F47fMJllTQWyGNa5hNROhjD7yRGsLTXfZgLF9CYkxUevrxllvQMPF2FnXfU-3D>
>> © Intuit, Inc. All rights reserved.
>> Privacy
>> <http://links.notification.intuit.com/ls/click?upn=n1D1vNUf2DDfuFJ7P-2Bs2FAQAT1a-2BVCUFNSb4yU6w-2FVKWimp-2BacsID0u3qz7RS5uMo92Y_bQiQ5-2BGmXnymGnBlJujf8tkwj-2BB5QDQ7IdQrLCC6j4u57rNnTY0XSZIlfrED7bMJEvsf4ZBvsN-2B-2Ff7gED6-2BE1iq6BDzNUH0CaI9-2FJ6-2BfpYlQ6UFYctK37YOfP31QEB7uB-2FsNC-2FkjHtugpiUzwdaQc7iRp7k-2FAhuAPgOV5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXIUCd4WCbpBu330BP3ff4df-2Bmezn0I7IKlSpyfKphRBCppS0l4sRUnSacfrLc7z6Pn5zNXsvrjxvI9zw9oCBsDHQR7T5y8y6-2Bdl0h60C-2BPZKotZcZMVwu-2B-2Fd0h5UfITz9z-2BK-2FaBIDxo0pAmCDjY8-2Fif6vtg-2BDfVBJ9X24K33af9FXpX3Njw40HLO2r79NaE3BD-2FZfnBqhR3DItJE1avG-2FjVB1ErS8b1H5fKX9Unp8bqqLlaOoYb627R8NT1kcfPmKxT9G3dgkMu2f0fXqMLQFBogmZevGgNo0b3QBXwFaTEea-2FZiz0xTtBjWLCsHNFMwNar-2F5LU-2BRpUCTHO-2F31E69EcTp1sjuhnYHOCM-3D>
>> | Security
>> <http://links.notification.intuit.com/ls/click?upn=0HF4wnohX-2BOI-2Bf-2Br3N3o2y8arzP1knvc61bgocVbBS-2FCwaVsGANJFSobkPia6WIwFzmKC8uL3IamkR1nM0Ne5Q-3D-3DbGAY_bQiQ5-2BGmXnymGnBlJujf8tkwj-2BB5QDQ7IdQrLCC6j4u57rNnTY0XSZIlfrED7bMJEvsf4ZBvsN-2B-2Ff7gED6-2BE1iq6BDzNUH0CaI9-2FJ6-2BfpYlQ6UFYctK37YOfP31QEB7uB-2FsNC-2FkjHtugpiUzwdaQc7iRp7k-2FAhuAPgOV5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXI [... cut ...] V5GOcH0caUHphw9-2FiWPbc6jpQjriRlIbrqGl7f3rTPDw4OyWE0aqnPm5WVQamxKsh9RAPdWMNRXD-2Ft5ixnJAWy5l2X-2BQJTr0TSJqXIUCd4WCbpBu330BP3ff4df-2Bmezn0I7IKlSpyfKphRBCppS0l4sRUnSacfrLc7z6Pn5zNXsvrjxvI9zw9oCBsDHQR7T5y8y6-2Bdl0h60C-2BPZKotZcZMVwu-2B-2Fd0h5UfITz9z-2BK-2FaBIDxo0pAmCDjY8-2Fif6vtg-2BDfVBJ9X2-2FHa366W4UrZDXHdzd2VjeCoe4qItpEM1znjI0nSPEk5qi7uB5g3GvnPqkrFUQ57KhSYwD0RKDak-2F6RnSatEjMbGA87MgGM9yKGMS95DBy8NG7JELYFRxpJfPKXk37sD1kQ7Gfx8Wk5HDe1QYKCXxVdQQx5gVzFtC280-2BD-2F5x3x8n2n8N0EXx7VCCsNjCeq3A-3D>
>>
>>
>