uavpal/disco4g #98: 4G internet shared over WiFi

wangxiaoming1986, 2021-08-04:
I apologise if this has already been discussed in previous threads, but, would it be possible to share the 4G internet over wifi on a bebop2 or disco? This would allow the internet access of various wifi sensors and devices connected to the drone while it is in flight. Could this be implemented at all?

Thanks

s0ftice, 2021-08-05:
Theoretically yes, but needs code changes.

wangxiaoming1986, 2021-08-05:
Understood :) Like with everything else, is it safe to assume that it will only be implemented if more users ask for it?

s0ftice, 2021-08-06:
either that or otherwise anyone can take the code and make changes to it, it's all open source :)

wangxiaoming1986, 2021-08-06:
Well, the best way to summarise it is: "Be the change you want to see in the world". So I might as well try and implement this function myself if I need it that much, rather than wait for others to do it for me. I may end up with a bricked drone in the process, but that's the risk when venturing on paths unknown ;)

s0ftice, 2021-08-06:
just wondering, what are your use cases? what wifi sensors do you intend to use?

wangxiaoming1986, 2021-08-06:
Well, one application would be to monitor my solar panels for hot spots with a wireless infrared camera. It's light enough for the drone to carry (tested it already), and accessing it in real time, over the internet, would be wonderful to be able to do without having to buy dedicated equipment just to do that. Bandwidth should not be an issue with these cameras, they are low resolution anyway.

Another application would be to install a microphone/speaker combo on the drone, so that I can speak to and hear from somebody "at a distance" :) This may not be practically feasible as the blades' noise would probably drown any communication. But I'd give it a try just to see how well it works, if at all. Failing that, I would like to control a high pitch speaker and play a police siren sound when visiting a mountain cabin that I own and which has been broken into several times already. In the winter the same cabin gets visited by bears and sometimes wolves, and I did notice that ultrasounds scare these "scary" animals away. So installing an ultrasound speaker which I can switch on and off as I please, would help me keep these animals away. It's not nice when you hear a bear in the winter scratching his bottom up your door and not knowing how to get rid of it.  And when, and if, somebody rents out the cabin and complains about bears, I would like to deploy the drone and bathe these unfortunate animals in ultrasounds. 

These are just a few scenarios I have in mind at the moment. If they work well, and with more WiFi devices coming on the market, I'm sure there will be more people wanting to connect all kind of things to their drone, and also be able to access them remotely. :)

So that's why I was wondering if anybody had tried or needed 4G internet shared over WiFi. But it seems to me that this request has come a little bit ahead of its time. :)

s0ftice, 2021-08-09:
Hehe, sounds fun!

Probably easier to find sensor which you could attach via USB to CHUCK. Also solved the power problem, otherwise you'll need two batteries.
I suggest you join our community at https://uavpal.com/slack and present your ideas there. Pretty sure you will find some like-minded pilots/hackers.

closed 2021-08-09 by s0ftice