uavpal/disco4g #99: Disco not connecting...

steelnick, 2021-09-10:
Hey guys,
for several days I have been trying to soft mode my disco, but it is not working. I have huaweii e3372h-153 (written on the modem), but apparently working as a stick-mode (firmware 21.x)
In the options menu I saw the apn, no passwords or pins required. On a pc its is working, but not on the parrot. I have downgraded to firware version 1.7.0, but there is no difference. 
I have a question - in telnet mode, when I copy all the lines into the command prompt, they are starting to execute, and finally I have to press enter to reboot, when do I have to turn of the skycontroller? Before hitting enter, while the disco is rebooting, or after the reboot?

Thx in advance!

Nick

ClassAxion, 2021-09-11:
"h" and "s" suffix is related to hardware version of the modem, not hilink / stick firmware.

s0ftice, 2021-09-13:
it doesn't matter at what time you turn on the skycontroller.
when you say in the options menu on the PC you can see the APN, what software is that? Is it "Mobile Partner"?

you can try to post some debug output in here, see https://uavpal.com/disco/faq#debug

steelnick, 2021-09-13:
Yes, its a software from the mobile partner (see the pic below)

[image]

When I run ulogcat, I am getting this. Unfortunately I cannot translate it :D (sorry, I am a monkey)
[image]

s0ftice, 2021-09-13:
1. Have you set the APN config file of the softmod to the same value? (internet.vivacom.bg)?
2. Do you know if the Username/password is mandatory by your carrier?
3. Not just `ulogcat`, please run `ulogcat |grep uavpal` as stated in the FAQ article I linked earlier.

steelnick, 2021-09-13:
1. Yes, I had
2. there is such option, but it is not activated
3. It didn't work the first time I tried, now I did it again:
[image]

steelnick, 2021-09-13:
One more thing - I have tried to create debug file, but obviously something goes wrong.
How am I supposed to create empty file, do I use notepad and then upload it by filezilla?

s0ftice, 2021-09-13:
no ignore the debug log file.

2. it looks activated on your screenshot. pls check with your carrier
3. that's still without `| grep uavpal` ...

steelnick, 2021-09-13:
2. ОК
3. hmm, I copied it from Your message, will try again...
When I hit enter, nothing happens
[image]

steelnick, 2021-09-14:
Nothing happens :(
Yes, the modem has led, when plugged in at first its off, after 3-5 sec it blinks with green couple of times, then cyan for 2 sec

steelnick, 2021-09-14:
thinking about updating to latest firmware

steelnick, 2021-09-14:
I installed the latest firmware and tried again... This time on ulogcat only I got this
[image]
Nothing else changed

s0ftice, 2021-09-14:
this is the entire extract, which does not help... without `| grep uavpal` please ...
and if that's still empty 30 seconds after plugging in the modem, you probably did not install the mod correctly.

Are you coping the command lines one by one or as one big block?

steelnick, 2021-09-14:
I reinstalled it, as follows:
1. telnet 192.168.42.1, enter
2. mv /data/ftp/internal_000/disco4g-* /tmp/disco4g, enter, gives new line and "/ #"
3. chmod +x /tmp/disco4g/*/*_install.sh, enter, gives new line and "/ #"
4. /tmp/disco4g/disco/disco_install.sh, enter, starts installing, at the end gives new line and "/ #"
5. /tmp/disco4g/skycontroller2/skycontroller2_install.sh, starts installing, at first is trying to connect, not connecting, tries again, connected, installing, at the end gives new line and "/ #"
6. reboot, enter
7. after reboot is done (skycontroller led is green) I am shutting off the controler. USB modem is flashing cyan every 2 sec, connecting, telnet ulogcat | grep uavpal - nothing :( 
Sorry for being a monkey, but - on windows "|" is one symbol, but in command prompt is like ":", I have tried with copying and the symbol changes.

steelnick, 2021-09-15:
Here are some pictures from install:
[image]
[image]

s0ftice, 2021-09-16:
Installation looks fine. The pipe symbol looks like a colon in Windows, that's also fine. Are you plugging the modem into the right or left USB port (looking from behind at CHUCK)? should be right. what output does `lsusb` show? (run in a telnet session about 10 second after the modem is plugged in)

What could be: your USB cable is a power cable only without data wires.

steelnick, 2021-09-16:
Everything is as You have written, the cable has to be OTG, haven't tried it though.
1susb showed:
[image]

s0ftice, 2021-09-16:
I'm not at home currently to compare, can you run `lsusb` when the modem is not plugged in? Does it show the same 4 devices?

steelnick, 2021-09-16:
Nope
[image]

s0ftice, 2021-09-16:
lsusb, not 1usb!

steelnick, 2021-09-16:
Sorry, my bad! The devil is in the details! :)
Looks pretty much the same :(
[image]
So, the problem is with the usb, too bad I have payed 4 euros for that "OTG" cable, will try to solder one tomorrow, any recommendations?

s0ftice, 2021-09-16:
Yes, simple answer: your cable is either broken or does not have data wires at all. Try a different cable.

steelnick, 2021-09-17:
I just soldered a new OTG cable, but I am getting the same answer. Is it possible the usb port of CHUCK to be the problem? Should I open it to check it, or there is another way?

s0ftice, 2021-09-17:
you can try the other port, check https://uavpal.com/disco/wiki/How-to-fix-a-broken-Micro-USB-port-on-CHUCK (Option 2)
but you also have to make a special cable.

steelnick, 2021-09-17:
OOOOK! Finally worked! After the start I got almost right away steady cyan :) 
Unfortunately now I have another problem - not connecting to zerotier, on the members tab shows no devices has joined

steelnick, 2021-09-17:
I uninstalled the mod, installed it again, nothing changed :( am I loosing something on the way? On skycontroller everything worked fine - connected to zerotier. What could be the problem with the disco?
s0ftice, thank You for the help! I really appreciate it!

steelnick, 2021-09-20:
I tried to reinstall the mod only with zerotier id, same problem - controller is connecting but the disco is not.
Ran the ulogcat|grep uavpal:
[image]

s0ftice, 2021-09-20:
the modem seem connecting and disconnecting non-stop. Maybe the cable has a loose connection?

steelnick, 2021-09-20:
the modem light is not blinking, may be is not a power issue. Will double check! Thx again :)

steelnick, 2021-09-20:
the OTG shortening seemed a bit loose, I rewelded it.
[image]

s0ftice, 2021-09-21:
I'm not sure. You could try to flash a Hilink firmware to the modem.