uavpal/disco4g #103: Connection failure with Huawei MS2372h-517 Modem

Oliver1948, 2021-11-09:
Hey everyone!
I have recently tried to use the 4g mod on my drone with a Huawei MS2372h-517 modem, but it doesn't want to connect. When I look in the ulogcat, the drone recognizes the modem, and then tries to identify it as Stick or Hi-link mode (Detecting Huawei Modem Type). That is the last message displayed about my modem, and nothing else happens. A few people have recommended installing new firmware on the modem, but I don't know where I would find new firmware, and I don't know how to do it, as my modem gives me no way to configure its settings. If anyone has any experience with this modem, I would greatly appreciate their advice.

Thanks in advance,
Oliver

[image]

s0ftice, 2021-11-10:
did you swap the 2 variables as advised here: https://uavpal.com/disco/faq#othermodems

:point_right: for the E3276s and MS2372h, you need to swap the values (0 and 1) of the variables serial_ctrl_dev and serial_ppp_dev in /data/ftp/uavpal/bin/uavpal_disco.sh.

Oliver1948, 2021-11-10:
Yes I did, but nothing changed.

s0ftice, 2021-11-10:
nothing happens after the last "detecting huawei modem type".
maybe you run `ulogcat |grep uavpal` to filter out the noise.

Oliver1948, 2021-11-10:
I did what you said, but still the last thing that poped up regarding the modem was
I uavpal_drone(ulogger)                        : ... detecting Huawei modem

Oliver1948, 2021-11-14:
Could I need new firmware?

s0ftice, 2021-11-17:
Yes, I would probably try a different firmware then. Don't ask for links :)

Oliver1948, 2021-11-17:
I found the solution! Buy a new modem ;)

closed 2022-02-12 by s0ftice