uavpal/disco4g #105: Connection Failure

dtfoz, 2021-12-15:
Hoping someone can help. I have just finished the conversion and have the Skycontroller talking via web fine to zerotier. Im running firmware 1.7.1 and have a E3372h-608 modem which sometimes shows a solid blue light. Even when this happens, it does not register with Zerotier. I have a proper OTG cable, have tried removing and reinstalling the mod to no avail. Please help :(

s0ftice, 2021-12-15:
the E3372h-608 is not compatible with the mod.

dtfoz, 2021-12-15:
Can i just confirm E3372h-607 will work s0ftice?

s0ftice, 2021-12-15:
yes it will. Just make sure that the supported LTE bands (1,3,7,8,28,40 for the -607 model) is supported by your mobile operator.

dtfoz, 2021-12-23:
So....went and purchased a new modem which is **E3372h-153 3G/4G/LTE USB Modem with Wi-Fi Fully Unlocked**

New carrier (Telstra) and works auto when plugged into the pc. Still only receiving connection from Skycontroller on ZeroTier. Uninstalled, reinstalled, still nothing. Even went as far as changing out the OTG cable and still nothing from the Disco :( Pleeeeaasseee help.....

s0ftice, 2022-01-07:
can you post the output of `ulogcat |grep uavpal` ?
run the command in a telnet session on the disco 1-2 minutes after you plugged in the modem (with the softmod being installed)

dtfoz, 2022-01-14:
Finally have it now up and running! Had to go order an E3372-607H which worked pretty much the second i plugged it in! Thanks @s0ftice for your help.

closed 2022-01-15 by s0ftice