uavpal/disco4g #107: E3372s-153 Bricked?

Crussade, 2022-06-03:
Hello,

Today I tried to install the softmod and followed the instructions. As I plugged in the E3372s i got to the internet page and was able to configure everything and get a internet connection. plugged the stick to the disco and started installing. Now the stick always reebots after a few seconds. if i put it in pc he got recognized and the usb instantly disconnects again and is stuck in this loop forever. is there a way to save my E3372s

s0ftice, 2022-06-07:
the softmod won't brick your stick. can you try on a different PC just to make sure.

Crussade, 2022-06-07:
I already did, same behavior. I found out i used the wrong USB Port on the Disco, didn't know at that time it had no 5v power. So I think the stick is toast :(.

s0ftice, 2022-06-07:
That's quite surprising to hear. The left port is also 5V, it's just underpowered in terms of amps. That should not damage the modem, it will just not work on that particular port. Quite a few pilots broke off their right port during a crash with the modem plugged in. And pretty much everyone tried the left port before finding my instructions here: https://uavpal.com/disco/faq#brokenport

But nobody else has reported that the modem got permanently damaged after plugging into the left ports so far...

s0ftice, 2022-06-08:
you can try to re-flash/upgrade the firmware. I close the issue for now, as it's not directly related to the softmod

closed 2022-06-08 by s0ftice