uavpal/disco4g #110: Issue glympse link

Fred06220, 2022-12-15:
Hi, how are you? I'm Fred from France.
First of all, thank you for the job you've provided. You really did a good work. Thank you very much.
 well my problem is : the Install worked, but the link I receive from glympse, lead me to the internet site not to the app. Instead receiving https://glympse.com/06UY-JH5N (example),
I receive https://glympse.com.

I've checked all the process I had to do and I did the job well, but I can't receive the good glympse's link to track my Disco. my 4g key works perfectly and the disco's control finish to be connected to the Disco (blue light stay fix). Does anyone could help me please? 
I  attached file, I put a screen shot of pushbullet.

[image]
 Thank you!
Fred.

s0ftice, 2022-12-15:
Glympse should be working again in the latest beta (see details here: https://uavpal.com/disco/faq#beta). Or simply replace the uavpal_glympse.sh file using https://raw.githubusercontent.com/uavpal/disco4g/master/disco/uavpal/bin/uavpal_glympse.sh

closed 2022-12-15 by s0ftice

Fred06220, 2022-12-15:
Thank you very much. I'm gonna try it. I'll let you know if it works.
Thank you again