uavpal/disco4g #114: !HELP!  https://developer.glympse.com/account/create

DarionFlys, 2024-09-30:
Since this link is now down..I'm not sure what to do. Can anyone offer me some up to date help or alternatives? I am eager to start my 4G mod 10 years late, but I am stuck at step 1.. Thanks!!

https://developer.glympse.com/account/create

s0ftice, 2024-10-01:
Following screenshot taken from https://www.facebook.com/groups/173527569745090/permalink/1939831409781355/
You can reach out to Cami Zimmer to get an API key.

[image]

keep in mind, Glympse is optional. You could also do the mod without it.

closed 2024-10-01 by s0ftice

DarionFlys, 2024-10-01:
Softice you are a real lifer saver thank you. I really just want to do the
mod and be able to have telemetry information. Thank you very much.