uavpal/disco4g #116: Not really an issue - ZeroTier

Xprogrammer777, 2025-01-20:
I don't really understands how ZeroTier works, why it doesn't require any account. 
I thought it was some kind of tunneling like Ngrok, if it's the case that mean the connection is routed in the server of the repo owner? Someone said it doesn't use any server so I'm really confused.

s0ftice, 2025-01-21:
It does require an account.

This is how it works:
if it can establish a direct connection (e.g. CGNAT on both sides), no external server is used to route the traffic through.
if it can't, it needs to use a relayed connection, where it routes the traffic (encrypted) through zerotier's relay servers (not the repo owner's).
You also have the option to set-up your own relay server (called zerotier moon).

Xprogrammer777, 2025-01-21:
> if it can establish a direct connection (e.g. CGNAT on both sides), no external server is used to route the traffic through.

STUN? Hole punching? I still don't understand how direct connection work

s0ftice, 2025-01-22:
zerotier uses a combination of several techniques, yes UDP hole punching being part of it. you can read more about it here https://www.zerotier.com/blog/the-state-of-nat-traversal/

closed 2025-08-02 by Xprogrammer777