uavpal/disco4g #117: Glympse not sending unique URL

Tuttle13, 2025-02-19:
Hi, I note there was previously a similar issue, but I am finding that when I receive the Glympse SMS to my phone, it only says:

You can track the location of
your DISCO-050706 here:
https://glympse.com/

The link above is all that is sent in terms of a URL. So clicking on it just takes me to the homepage of Glympse. 
I set my Glympse account up 2+ years ago, and successfully got a API key (which is onboard the Disco).

I have just now tried installing the most up to date version of uavpal_glympse.sh (the 4 year old version) but I’m not not even getting this text from above, only one text about my SD card size and available space? 

Very confused, I’ve checked my uavpal_glympse file and it’s definitely including my API key on the top line?

s0ftice, 2025-02-19:
try this first please https://github.com/uavpal/disco4g/issues/110

Tuttle13, 2025-02-19:
@s0ftice thats the file I used to replace the Glympse file with:

https://raw.githubusercontent.com/uavpal/disco4g/master/disco/uavpal/bin/uavpal_glympse.sh

Tuttle13, 2025-02-20:
No, I mean I have both confirmed that the api is provided in its file (glympse_apikey) and that the uavpal_glympse file ( untouched) has been downloaded from that link above, and replaced the identical named file onboard the Disco

s0ftice, 2025-02-20:
you can try to run `ulogcat | grep uavpal` before you plug in the modem.
once plugged it, wait 1-2 minute and see if there's any relevant output.

My best guess: it's either a wrong API key or you did not replaced the uavpal_glympse script properly (maybe wrong location?)

josefede, 2025-05-04:
@Tuttle13 make sure you haven't changed the file's permissions. If you deleted the file on the Disco and uploaded the updated one that's likely the case, and you not receiving any new SMSs after reverting to the old file further suggests this. For anyone trying to modify this file, I recommend directly editing uavpal_glympse.sh on FileZilla to avoid this.

To fix your issue, double press the Pitot tube on the Disco after booting to enable Telnet (you should see a purple light). Connect to the Disco's hotspot and open a cmd window or terminal, depending on your OS.

Run the following commands:
```
telnet 192.168.42.1
chmod 750 /data/ftp/uavpal/bin/uavpal_glympse.sh
exit
```
You now should have the proper permissions on that file and it should run just fine. I can confirm that replacing the file with the one s0ftice provided makes the Disco send the full Glympse link.

Hope this helps.

josefede, 2025-05-04:
@s0ftice maybe time to create a new release so that people don't keep downloading the old Glympse file? The link [](https://uavpal.com/download-latest-disco) that is included in the installation section still has the old one